adrianmihalko/raspberrypiwireguard #1: small question about explanation

hetzbh, 2017-11-20:
In your instructions connecting both Raspberry Pi A & B, shouldn't Raspberry Pi A have rpia_public.key and Pi B have rpib_public.key ?

adrianmihalko, 2017-11-20:
No, On Raspberry Pi A you must use Raspberry Pi B's key (rpib_public.key) and vice versa. 

Then if you connect with Raspberry Pi A to Raspberry Pi B: Raspberry Pi B can pair/sign his (rpib_public.key  from the config on Rpi A) with his private key. And vice versa too.

hetzbh, 2017-11-20:
Oh, you're right, but what happens when I want to connect between >2 Pi's? (5, for example)

adrianmihalko, 2017-11-20:
Each WireGuard interface may have multiple peers:

           [Interface]
           PrivateKey = yAnz5TF+lXXJte14tji3zlMNq+hd2rYUIgJBgB3fBmk=
           ListenPort = 51820

           [Peer]
           PublicKey = rpib_1.publickey
           Endpoint = 192.95.5.67:1234
           AllowedIPs = 10.192.122.3/32, 10.192.124.1/24

           [Peer]
           PublicKey = rpic.publickey
           Endpoint = [2607:5300:60:6b0::c05f:543]:2468
           AllowedIPs = 10.192.122.4/32, 192.168.0.0/16

           [Peer]
           PublicKey = rpid.publickey
           Endpoint = test.wireguard.io:18981
           AllowedIPs = 10.10.10.230/32

...etc. Feel free to ask at: [#Wireguard](http://webchat.freenode.net?randomnick=1&channels=%23wireguard&uio=d4)

hetzbh, 2017-11-20:
Thanks for the explanation ;)

adrianmihalko, 2017-11-20:
No problem, use it, it's a great tool!

closed 2017-11-20 by adrianmihalko