adrianmihalko/raspberrypiwireguard #4: Better way to install for Raspberry Pi 2+

ViRb3, 2018-07-01:
Following the official WireGuard Debian instructions:

```
# apt install raspberrypi-kernel-headers -y # required to build kernel module
# echo "deb http://deb.debian.org/debian/ unstable main" > /etc/apt/sources.list.d/unstable.list
# printf 'Package: *\nPin: release a=unstable\nPin-Priority: 150\n' > /etc/apt/preferences.d/limit-unstable
# apt update
# apt install wireguard
```

adrianmihalko, 2018-09-04:
I get:

pi@raspberrypi:~ $ sudo wg
Segmentation fault

...using this method.

ViRb3, 2018-09-04:
Are you sure you are running a Pi 2/3? The older Pi's are missing you extensions.

adrianmihalko, 2018-09-04:
I tried on RPi Zero (non Wlan). 

+ one more info I was needed to install 
$ sudo apt-get install dirmngr

to import keys for APT.

ViRb3, 2018-09-04:
For the Pi Zero follow this:
https://github.com/adrianmihalko/raspberrypiwireguard/issues/3

adrianmihalko, 2018-12-19:
updated guide

closed 2018-12-19 by adrianmihalko