adrianmihalko/raspberrypiwireguard #6: Segmentation fault

furrybanana, 2018-12-21:
Hi there,

tried to follow your Tutorial but stuck on generating keys on a raspberry pi 2.

pi@raspberrypi:~/wgkeys $ wg genkey > server_private.key
Segmentation fault

Syd, 2018-12-21:
also segfaulting on a raspberry pi 1. From memory they're the same architecture?

adrianmihalko, 2018-12-21:
Yep, Pi Zero/ 1 /2 are the same, except the Pi 2 v1.2(it was released in late 2016).

furrybanana, 2018-12-21:
Hi and thank you for the fast answer.

the same microsd card works in a raspberry pi 3. I could generate keys in an instant. 

i did apply the command before and it returned:
net.ipv4.ip_forward=0

i hat to edit **#**net.ipv4.ip_forward=**0**  --> net.ipv4.ip_forward=1

adrianmihalko, 2018-12-21:
Did you rebooted too? You should. But maybe I am going to remove this from guide, because manual method is the best.

furrybanana, 2018-12-21:
yes i always rebooted like mentioned in the tutorial.

adrianmihalko, 2018-12-21:
Thanks, I try to reproduce and fix then.

Syd, 2018-12-21:
Seems the issue is the build is for ARMv8, might need to upgrade the guide to recommend a 2 v1.2 or greater.

furrybanana, 2018-12-21:
Thank you, i will try another Pi 2, as i got all of them (expect the newest 3).

adrianmihalko, 2018-12-21:
Please run on fresh install:

`pi@raspberrypi:~ $ sudo sed -r 's/#{1,}?net.ipv4.ip_forward ?= ?(0|1)/net.ipv4.ip_forward = 1/g' /etc/sysctl.conf  
`
It should not rewrite the file, but show the result in terminal. So we can check if it has been replaced.

@Syd Done. If there is an other way to install on these older architectures, let me know. I'll try to find other method.

furrybanana, 2018-12-21:
so i tested another pi 2 v1.1 and it works. don't know why one is not working. same sdcard, power supply, network cable, genkey now works.

maybe that pi it is dead or something went corrupt.

for fun i tried a Pi Model B+ V1.2 (2014) - throws Segmentation fault 

so confirmed working on:

Pi 2 Model B V1.1 (2014) 
Pi 3 Model B V1.2 (2015)

adrianmihalko, 2018-12-21:
Hmm, that's weird.

What

`pi@raspberrypi ~ $ cat /proc/cpuinfo`

is showing. It is same for both pi 2?

My conclusion: 2B should work, because of ARMv7 architecture?

ARMv6 (1, Zero, ZeroW) pretty sure needs manual compiling.

The table of architecture is here:

https://en.wikipedia.org/wiki/Raspberry_Pi
---

I updated the guide, no need for rpi-source:

https://github.com/adrianmihalko/raspberrypiwireguard/wiki/Install-WireGuard-on-Raspberry-Pi-1,-2-(not-v1.2),-Zero,-Zero-W

It's actually pretty simple to compile manually. Yes, upgrade requires one more step, like a simple apt-get update/upgrade.

furrybanana, 2018-12-21:
> Please run on fresh install:
>
> `pi@raspberrypi:~ $ sudo sed -r 's/#{1,}?net.ipv4.ip_forward ?= ?(0|1)/net.ipv4.ip_forward = 1/g' /etc/sysctl.conf `
> It should not rewrite the file, but show the result [...]

so after 

pi@raspberrypi:~ $ sudo sed -r 's/#{1,}?net.ipv4.ip_forward ?= ?(0|1)/net.ipv4.ip_forward = 1/g' /etc/sysctl.conf 

and reboot, I get: 
pi@raspberrypi:~ $ sudo sed -r 's/#{1,}?net.ipv4.ip_forward ?= ?(0|1)/net.ipv4.ip_forward = 1/g' /etc/sysctl.conf

furrybanana, 2018-12-21:
shoot, already manual'd my way. will try on another pi later.

furrybanana, 2018-12-21:
Nope, still 
`sysctl net.ipv4.ip_forward 
net.ipv4.ip_forward = 0
`

adrianmihalko, 2018-12-21:
Good question, what is happening here. I try to do more research.

furrybanana, 2018-12-21:
Other question:

Do I have to set an Endpoint in my server wg0.conf ? 

> [Interface]
**Endpoint = myDNSadress.com:51820**
Address = 10.0.1.1/24
ListenPort = 51820

PrivateKey = KEYKEY
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -A FORWARD -o %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -D FORWARD -o %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE

[Peer]
#Client1
PublicKey = YEKYEK
AllowedIPs = 10.0.1.2/32

adrianmihalko, 2018-12-21:
No, it just for clients.

adrianmihalko, 2018-12-21:
I was able to reproduce bug, this should work

`sudo perl -pi -e 's/#{1,}?net.ipv4.ip_forward ?= ?(0|1)/net.ipv4.ip_forward = 1/g' /etc/sysctl.conf`

Starting key was:
#net.ipv4.ip_forward=0

furrybanana, 2018-12-21:
Thank you, will try it again later.

i am trying to get this to run on a rapsberry pi 1 with this link: https://github.com/adrianmihalko/raspberrypiwireguard/wiki/Install-WireGuard-on-Raspberry-Pi-1,-2-(not-v1.2),-Zero,-Zero-W
i could generate keys but cant connect. do i have to do the other stuff too? 

> pi@raspberrypi:~ $ echo "deb http://deb.debian.org/debian/ unstable main" | sudo tee --append /etc/apt/sources.list.d/unstable.list
pi@raspberrypi:~ $ sudo apt-get install dirmngr 
pi@raspberrypi:~ $ sudo apt-key adv --keyserver   keyserver.ubuntu.com --recv-keys 8B48AD6246925553 
pi@raspberrypi:~ $ printf 'Package: *\nPin: release a=unstable\nPin-Priority: 150\n' | sudo tee --append /etc/apt/preferences.d/limit-unstable

adrianmihalko, 2018-12-21:
No, you do not need this after you compiled manually. After you compiled, verify it works by running `sudo wg`. You should not get any error after that.

Then your next step is to start configuring.

**2. Configuring WireGuard** and the rest.

furrybanana, 2018-12-21:
ok did that, i have setup server, client and ios but i fail to ping the server from client.

adrianmihalko, 2018-12-21:
BTW check this guide too, it helps a lot in configuring multiple users, mobile clients:

furrybanana, 2018-12-21:
I wanted to thank you again. Great work.

Now i only need to find out, why it does only work if Allowed IPs is set to 0.0.0.0/0 - can't get my own working

adrianmihalko, 2018-12-21:
In mobile? If you set 0.0.0.0/0 you should fill DNS too to 8.8.8.8. If it still not working, you can always ask help in Wireguard IRC channel:

http://webchat.freenode.net?randomnick=1&channels=%23wireguard&uio=d4

furrybanana, 2018-12-21:
yes ios and android only works with 0.0.0.0/0

thank you again :+1:

adrianmihalko, 2018-12-21:
Oh I see now, sorry. What's your local subnet? In the example I showed config for 192.168.1.x

With the setup from the example you can access devices in the 192.168.1.x range + the WG server at 192.168.99.1.

furrybanana, 2018-12-21:
router 192.168.178.1
wireguard server/subnet 10.0.0.1

on mobile i could get to my printer, but it won't work with printername.local:port

adrianmihalko, 2018-12-21:
Oh, good question, but I don't think you can use hostnames on VPN. Ask it for sure on IRC, I am not sure in this.

furrybanana, 2018-12-21:
> I was able to reproduce bug, this should work
> 
> `sudo perl -pi -e 's/#{1,}?net.ipv4.ip_forward ?= ?(0|1)/net.ipv4.ip_forward = 1/g' /etc/sysctl.conf`
> 
> Starting key was:
> #net.ipv4.ip_forward=0

can confirm, works now.
can be closed, went way to offtopic. thank you very much @adrianmihalko :1st_place_medal:

adrianmihalko, 2018-12-21:
Cool, thank you.

closed 2018-12-21 by adrianmihalko