adrianmihalko/raspberrypiwireguard #8: How about 2 conf files for each client

ljm42, 2018-12-27:
Thank you for these instructions! The wg_config project is helpful as well. What do you think of expanding the wg_config script so that it creates two .conf files for each client:
* clientSplit.conf - with the split tunnel defined in client.conf.tpl
* clientAll.conf - with AllowedIPs set to 0.0.0.0/0
Clients can install both and then easily choose which they want to use depending on whether they trust their current network.

adrianmihalko, 2018-12-28:
Done. `name.all.png` and  `name.all.conf`  are created.

ljm42, 2018-12-28:
That looks great, thanks!

What do you think about adding a "view user split" option, perhaps as:
```
 ./user.sh -s client1
```
that would do something like this:
```
cat $userdir/client.conf
qrencode -t ansiutf8  < $userdir/client.conf
```

Similarly, "view user all":
```
 ./user.sh -a client1
```
would do the same but on the client.all.conf file:
```
cat $userdir/client.all.conf
qrencode -t ansiutf8  < $userdir/client.all.conf
```

The reason is that I really like how you added `qrencode -t ansiutf8` to the script, but right now it is only available when adding a user. Trying to make it more accessible :) 

This could potentially regenerate the .png file too, not sure

adrianmihalko, 2018-12-28:
Done. Added `-v` option to view generated QR codes.

ljm42, 2018-12-28:
I love the qr codes :) You don't see a value in cat'ing the whole conf file though? I guess I can always cd to the user dir and cat it manually.

An observation... if you try to add a user that already exists, it overwrites the existing user in the users directory, but it appends the user to the .saved file. This means that both the old version of the user and the new version of the user are added to wg0.conf but you've lost the keys needed to make the original user work.

Can we have it replace the existing user entry in the .saved file rather than append a new one?

Taking it a step further, if the key files already exist in the users directory, maybe it shouldn't recreate them. In that case, "adding" an existing user would essentially mean "regenerate the conf files based on the latest "client.conf.tpl", without changing the keys or ips.

adrianmihalko, 2018-12-28:
Done. I keep it simple. -v shows not .conf file content. Added option to overwrite current user (it works like: delete, add). I think it is more than enough here.

ljm42, 2018-12-28:
delete/add is the perfect solution, thanks! It is interesting to realize that there is really no reason to save the client's old keys, just delete and re-create them. 

One thing I might suggest would be to call view_user at the end of add_user

And just for aesthetics, maybe sort the .available_ip file after returning the deleted user's ip?
   sort .available_ip --version-sort  > .available_ip

This script handles the "roadwarrior into a single server" use-case very nicely. What I am actually interested in doing is setting up site-to-site VPN between two Raspberry Pis, where users can be added at either end.  I think this would be doable by manually editing the .available_ip file so each server has its own range that it can give out, right? And I'd have to manually add each server as clients of the other. And then on the routers at each end, add a route through the Pi to the other network (unfortunately I only have access to one end of the tunnel right now so I can't really test it yet)

adrianmihalko, 2018-12-28:
Nice tip for `sort` command, I added at the end of delete command.

For the second question: I leave this question open as I don't have idea for it.

closed 2018-12-28 by adrianmihalko

ljm42, 2018-12-29:
Thanks for all of the help with this, it is a nice simple way to manage Wireguard. For now I'm not going to worry about site-to-site.