adrianmihalko/raspberrypiwireguard #11: Wireguard not starting because of wrong kernel-headers?

garret, 2019-01-05:
I was very happy to found out about your script on DrZZZ stream though I cannot get it working on my raspberry pi 3 B+. I am using dietpi as distribution. That is updated to the latest version. 

I followed the initial part (before choosing for manual or semiautomatic script) and everything seems fine (I get the blue messages you show). I then went for the semiautomatic script but I get this error: 

```
root@livingroom:~/wg_config# sudo wg-quick up wg0
[#] ip link add wg0 type wireguard
RTNETLINK answers: Operation not supported
Unable to access interface: Protocol not supported
[#] ip link delete dev wg0
Cannot find device "wg0"
root@livingroom:~/wg_config#
```

I found out that the wireguard module is not loaded:
```
root@livingroom:~# sudo modprobe wireguard && lsmod | grep wireguard
modprobe: FATAL: Module wireguard not found in directory /lib/modules/4.14.90-v7+
```

This is probably due to the kernel-headers installed do not correspond to the kernel I am using probably? 
```
root@livingroom:~# sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-4.14.90-v7
E: Couldn't find any package by glob 'linux-headers-4.14.90-v7'
E: Couldn't find any package by regex 'linux-headers-4.14.90-v7'
root@livingroom:~#
```

Do you know how I could fix this issue? Thank you very much in advance!

adrianmihalko, 2019-01-05:
We can try to debug this, but I think the result will be the same raspberrypi-kernel-headers is different version than the actual kernel. If you want to debug this, you can find me on Discord, but I think better choice is to change to raspbian, it is more wisely used distro. Or you can always try to install sources with rpi-source, I used this some months ago.

garret, 2019-01-06:
I am seeing that it should be possible to install wireguard on dietpi. I will try to ask support there since it deals with their distro. 

https://github.com/Fourdee/DietPi/issues/2052

adrianmihalko, 2019-01-06:
Cool, I hope you find a solution. If not, you can always contact me. I am closing this issue. :)

closed 2019-01-06 by adrianmihalko

joevo2, 2020-05-25:
Have the same issue :( on RPI 3 as well

Rohlik, 2020-05-28:
Also have this problem on RPi3+:
```
#apt-get install  "linux-headers-$(uname -r)"
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-4.19.114-v7
E: Couldn't find any package by glob 'linux-headers-4.19.114-v7'
E: Couldn't find any package by regex 'linux-headers-4.19.114-v7'
```

```
#uname -r
4.19.97-v7+
```

ukaeufe, 2020-06-01:
I updated raspbian and reinstalled wireguard and wireguard-dkms and it worked again after a reboot:

joevo2, 2020-06-02:
> I updated raspbian and reinstalled wireguard and wireguard-dkms and it worked again after a reboot:
> sudo -i
> apt-get upgrade
> apt-get update
> apt install --reinstall wireguard
> apt install --reinstall wireguard-dkms
> reboot
> 
> sudo wg show wg0
> 
> works again

Thanks! somehow it worked!

Amalinda, 2020-06-25:
similar issue on Arch, Linux 5.6.15-arch1-1 #1 SMP PREEMPT Wed, 27 May 2020 23:42:26 +0000 x86_64 GNU/Linux

Msprg, 2020-09-13:
For anyone struggling still with this issue, try:

```
sudo -s
apt update
apt install git bc bison flex libssl-dev -y
#This one will take a loong time
wget https://raw.githubusercontent.com/RPi-Distro/rpi-source/master/rpi-source -O /usr/local/bin/rpi-source && sudo chmod +x /usr/local/bin/rpi-source && /usr/local/bin/rpi-source -q --tag-update
rpi-source
apt remove wireguard wireguard-dkms wireguard-tools -y
reboot
```
lastly, after reboot:
`sudo apt install wireguard -y
`
I was running those on RPi0w. Yeah, big mistake, took like an hour to be done... But it works!

foxpup, 2020-11-20:
> For anyone struggling still with this issue, try:
> 
> ```
> sudo -s
> apt update
> apt install git bc bison flex libssl-dev -y
> #This one will take a loong time
> wget https://raw.githubusercontent.com/RPi-Distro/rpi-source/master/rpi-source -O /usr/local/bin/rpi-source && sudo chmod +x /usr/local/bin/rpi-source && /usr/local/bin/rpi-source -q --tag-update
> rpi-source
> apt remove wireguard wireguard-dkms wireguard-tools -y
> reboot
> ```
> 
> lastly, after reboot:
> 
> ```
> #This might take just forever also...
> sudo apt install wireguard -y
> ```
> 
> I was running those on RPi0w. Yeah, big mistake, took like an hour to be done... But it works!

Thanx.  That really helped me out.  LLAP