adrianmihalko/raspberrypiwireguard #13: Raspbian: new kernel - wireguard not working

codegrau, 2019-02-21:
Hi there. Raspian updated to new kernel (and kernel headers) but wireguard won't start anymore

````
raspberrypi-kernel/stable,now 1.20190215-1 armhf [installed]
raspberrypi-kernel-headers/stable,now 1.20190215-1 armhf [installed]
````

````
# sudo modprobe wireguard
modprobe: FATAL: Module wireguard not found in directory /lib/modules/4.14.98-v7+
````

````
# sudo apt-get install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20190215-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
````

adrianmihalko, 2019-02-21:
Please try to install kernel sources manually and then reinstall Wireguard.

**0, Install required packages for rpi-source**

    sudo apt-get install bc libncurses5-dev

**1, Install kernel headers via rpi-source**

https://github.com/notro/rpi-source

codegrau, 2019-02-22:
Somehow this is solved now. Thanks!

dcaliskanQT, 2019-02-28:
Which kernel versions has to be installed? I installed rpi-source, what I have to follow now?

adrianmihalko, 2019-02-28:
Normally follow the guide on the main page. If you can't install Wireguard, because of kernel-headers are not found for your current kernel version, install kernel-source for your kernel version with rpi-source.

After that, try to install Wireguard again:

```
pi@raspberrypi:~ $ echo "deb http://deb.debian.org/debian/ unstable main" | sudo tee --append /etc/apt/sources.list.d/unstable.list
pi@raspberrypi:~ $ sudo apt-get install dirmngr 
pi@raspberrypi:~ $ sudo apt-key adv --keyserver   keyserver.ubuntu.com --recv-keys 8B48AD6246925553 
pi@raspberrypi:~ $ printf 'Package: *\nPin: release a=unstable\nPin-Priority: 150\n' | sudo tee --append /etc/apt/preferences.d/limit-unstable
pi@raspberrypi:~ $ sudo apt-get update
pi@raspberrypi:~ $ sudo apt-get install wireguard 
pi@raspberrypi:~ $ sudo reboot
```

If it still not working (or you have Rpi 1, Zero etc.), compile wireguard from source:

https://github.com/adrianmihalko/raspberrypiwireguard/wiki/Install-WireGuard-on-Raspberry-Pi-1,-2-(not-v1.2),-Zero,-Zero-W

WebMacke, 2019-04-03:
I think I have the same kernel version like @codegrau now with 4.14.

I have a new clean setup with raspbian stretch light on a rpi 1.
I encountered the problem with manual compiling

> 2, Clone WireGuard repository and compile it
> 
> ```
> pi@raspberrypi:~` $ git clone https://git.zx2c4.com/WireGuard
> pi@raspberrypi:~ $ cd WireGuard/src
> pi@raspberrypi:~/WireGuard/src $ make
> ```
>
This error occured:
```
make[1]: *** /lib/modules/4.14.79+/build: file or directory not found. Finish.
Makefile:37: the rule for target module failed
make: *** [module] error 2
```

I solved it with the hint from @adrianmihalko 

> 
> 
> Please try to install kernel sources manually and then reinstall Wireguard.
> 
> **0, Install required packages for rpi-source**
> 
> ```
> sudo apt-get install bc libncurses5-dev
> ```
> 
> **1, Install kernel headers via rpi-source**
> 
> https://github.com/notro/rpi-source

mbranchick, 2019-04-26:
Im having a similar issue.  I ran rpi-source, but Im still getting the following, any ideas?

pi@raspberrypi:~ $ modprobe wireguard
modprobe: FATAL: Module wireguard not found in directory /lib/modules/4.19.36-v7+

pi@raspberrypi:~ $ sudo apt-get install raspberrypi-kernel-headersReading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20190401-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

budius, 2019-05-01:
hi @mbranchick I was having same issues, I was also on 4.19, and core problem was that I used `rpi-update`, which weirdly udate to untested/unstable versions.

This rouce explains: https://www.raspberrypi.org/documentation/linux/kernel/updating.md
So I reverted back to 4.14 with:
```
sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel
```

and then with `sudo apt-get install raspberrypi-kernel-headers` everything matched and the kernel mdule compilation went fine.

leearmstrong, 2019-05-19:
Weirdly I get this too and for some reason following those steps gives me 4.19.42 now.  

Unpacking raspberrypi-kernel (1.20190517-1) over (1.20190517-1) ...

Have they updated to 4.19 now!?

andlil, 2019-06-01:
Easiest way to get around the missing module is to remove/reinstall wireguard-dkms

SomewhatDisoriented, 2019-08-09:
Running `dpkg-reconfigure wireguard-dkms` after each kernel upgrade is enough to fix this.

frasskungin, 2020-01-23:
> Please try to install kernel sources manually and then reinstall Wireguard.
> 
> **0, Install required packages for rpi-source**
> 
> ```
> sudo apt-get install bc libncurses5-dev
> ```
> 
> **1, Install kernel headers via rpi-source**
> 
> https://github.com/notro/rpi-source

I'm receiving error when running `sudo modprobe wireguard`, and I am assuming that it is due to the kernel issue:

> modprobe: FATAL: Module wireguard not found in directory /lib/modules/4.19.93-v7+

I didn't quite understand what is meant by "Install kernel headers via rpi-source". Could somebody elaborate on what the steps are for this?

Rohlik, 2020-05-28:
@frasskungin Here you find installation instructions - https://github.com/notro/rpi-source/wiki

ns408, 2020-05-29:
sudo tee -a /etc/rc.local <<EOF
if (modprobe wireguard 2> /dev/null); then
  dpkg-reconfigure wireguard-dkms || true
  modprobe wireguard || exit 1
fi
EOF

Raybo128, 2020-05-30:
Found that removing and reinstalling wireguard-dkms got me working again.  The dpkg-reconfigure command did not work for me.

`sudo apt remove wireguard-dkms wireguard-tools
sudo reboot
sudo install wireguard-dkms wireguard wireguard-tools
sudo reboot`

Afterward my wg0 interface returned and wireguard server was back up.

dallasgroot, 2020-06-03:
@Raybo128 Thanks! somehow this worked for me!

thisaccountname, 2020-06-22:
I tried most of the above. Ended up needing to run:

```
sudo -s
source /usr/src/wireguard-*/dkms.conf
dkms add wireguard/$PACKAGE_VERSION
dkms build wireguard/$PACKAGE_VERSION
dkms install wireguard/$PACKAGE_VERSION
```

reboot, re-start wireguard

mtnnn, 2020-07-25:
This command fixed it in my case:

` apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`