adrianmihalko/raspberrypiwireguard #15: installation problem?

jpgpi250, 2019-02-27:
1. wg0.conf: I tried to change the port from 51820 to something else, so I duplicated the line "ListenPort = 51820" and entered a comment sign (#) before the first entry. Wireguard ignores the comment and uses the first entry of "ListenPort = "

2. I installed Wireguard per instructions, tried both unstable and compiled instructions, using a fresh installation of Raspbian (latest - November 2018), NO errors on screen.
when I enter "sudo netstat -pln", I get the following:

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      510/sshd
tcp6       0      0 :::22                   :::*                    LISTEN      510/sshd
udp        0      0 0.0.0.0:68              0.0.0.0:*                           496/dhcpcd
udp        0      0 0.0.0.0:51820           0.0.0.0:*                           -
udp        0      0 0.0.0.0:52432           0.0.0.0:*                           337/avahi-daemon: r
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           337/avahi-daemon: r
udp6       0      0 :::38415                :::*                                337/avahi-daemon: r
udp6       0      0 :::51820                :::*                                -
udp6       0      0 :::5353                 :::*                                337/avahi-daemon: r
raw6       0      0 :::58                   :::*                    7           496/dhcpcd
Active UNIX domain sockets (only servers)
Proto RefCnt Flags       Type       State         I-Node   PID/Program name     Path
unix  2      [ ACC ]     STREAM     LISTENING     12298    496/dhcpcd           /var/run/dhcpcd.sock
unix  2      [ ACC ]     STREAM     LISTENING     12300    496/dhcpcd           /var/run/dhcpcd.unpriv.sock
unix  2      [ ACC ]     STREAM     LISTENING     7443     1/init               /run/systemd/private
unix  2      [ ACC ]     SEQPACKET  LISTENING     7456     1/init               /run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING     7467     1/init               /run/systemd/fsck.progress
unix  2      [ ACC ]     STREAM     LISTENING     7479     1/init               /run/systemd/journal/stdout
unix  2      [ ACC ]     STREAM     LISTENING     11852    589/systemd          /run/user/1000/systemd/private
unix  2      [ ACC ]     STREAM     LISTENING     11857    589/systemd          /run/user/1000/gnupg/S.gpg-agent.ssh
unix  2      [ ACC ]     STREAM     LISTENING     11860    589/systemd          /run/user/1000/gnupg/S.gpg-agent.extra
unix  2      [ ACC ]     STREAM     LISTENING     11862    589/systemd          /run/user/1000/gnupg/S.gpg-agent
unix  2      [ ACC ]     STREAM     LISTENING     11864    589/systemd          /run/user/1000/gnupg/S.gpg-agent.browser
unix  2      [ ACC ]     STREAM     LISTENING     10709    1/init               /var/run/avahi-daemon/socket
unix  2      [ ACC ]     STREAM     LISTENING     10712    1/init               /var/run/dbus/system_bus_socket
unix  2      [ ACC ]     STREAM     LISTENING     10715    1/init               /run/thd.socket

As you can see, port 51820 is only listed (NOT listening) on address 0.0.0.0 and :: 
Is this normal, what am I missing?

3. Is there an option to enable logging, to possibly see what is going wrong?

thanks for your time and effort.

adrianmihalko, 2019-02-27:
What does 

`sudo wg`

say?

jpgpi250, 2019-02-27:
sudo wg
interface: wg0
  public key: ********************************************
  private key: (hidden)
  listening port: 51820

peer: ********************************************
  allowed ips: 192.168.99.2/32

My wg0.conf:

[Interface]
Address = 192.168.99.1/24
ListenPort = 51820

PrivateKey = ********************************************
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -A FORWARD -o %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -D FORWARD -o %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE

[Peer]
#Client1
PublicKey = ********************************************
AllowedIPs = 192.168.99.2/32

adrianmihalko, 2019-02-27:
Please post your wg0.conf and be sure if you edit the correct file.

jpgpi250, 2019-02-27:
sudo cat /etc/wireguard/wg0.conf

[Interface]
Address = 192.168.99.1/24
ListenPort = 51820

PrivateKey = kEldXg/pnylu8cbZHoDlZpAERGa4+QIvABESDBwjR2k=
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -A FORWARD -o %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -D FORWARD -o %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE

[Peer]
#Client1
PublicKey = 90UAQlhWCyjguvuJXrrzLk/6w6wuXggqNYP0MO3Z8Rw=
AllowedIPs = 192.168.99.2/32

adrianmihalko, 2019-02-27:
I see no problem here and I can't see your uncommented Listenport line. Maybe you forgot to save the changes. So change Listenport to something else and restart the device, do not duplicate line.

jpgpi250, 2019-02-27:
1. the comment line has been removed, since it prevented me from changing the port, in order to try to debug. I'm just reporting this as a problem.
2. Logging?
3. The main problem that the configured IP is NOT listening on port 51820, e.g.
udp 0 0 0.0.0.0:51820 0.0.0.0:* -
udp6 0 0 :::51820 :::* -
I'm really wondering if the output is what I should expect.

adrianmihalko, 2019-02-27:
I don't really understand what is the problem. Did you tried connect to your WG instance?

2. No logging.

3. The output on my machine is:

```
@ubuntu:~$ sudo netstat -pln | grep 49926
udp        0      0 0.0.0.0:49926           0.0.0.0:*                           -
udp6       0      0 :::49926                :::*                                -
```

Exactly as yours.

1. double check keys
2. check if your port forwarding is OK, check your host/ip.

...and it should work.

If you still unable to connect to your Wireguard:

connect to Wireguard community on #Wireguard (IRC, Freenode).

jpgpi250, 2019-03-05:
This may be a stupid question, however, I can't resolve the problem (found a work around and got it working, but still would like to know why it doesn't work).

running Raspbian, latest version.

The IP of the eth0 interface is 192.168.1.57

On this machine, I've installed pihole, which is basically dnsmasq. dnsmasq provides DNS services on port 53 on the specified interface (you need to have `interface=eth0` in the configuration).

If I specify 192.168.1.57 (the eth0 interface) in the wireguard configuration, DNS resolving is NOT functioning.
If I specify another internal IP address (a second pihole) as the DNS server, everything works.

I assumed, since routing needs to be setup between the eth0 interface (192.168.1.57) and the wg0 interface (192.168.99.1), DNS resolution would work, however, it doesn't.

Of course, I configured `Allowed IPs : 192.168.99.1/32, 192.168.1.57/32` 

Work around: add `interface=wg0` to the dnsmasq configuration, and, wireguard configuration: `DNS=192.168.99.1` and `Allowed IPs: 192.168.99.1/32`

So the question: Why can I not use the eth0 IP address as the DNS server?

Thanks for your time and effort.

ThePigPeppa, 2019-09-20:
@jpgpi250 
 
**Automated Script:**

```
https://github.com/complexorganizations/wireguard-install
```