adrianmihalko/raspberrypiwireguard #17: Compile error on raspberry pi zero

modidep, 2019-03-17:
Hi,

I use Raspian image 2018-11-13-raspbian-stretch-lite

But when ever I try to compile I receive the below errors. I tried to compile with previous version of wireguard. It compiles OK but when you make wg-quick up wg0. It gives too few arguments after resolvconf.

` CC [M]  /home/pi/WireGuard/src/main.o
In file included from <command-line>:0:0:
/home/pi/WireGuard/src/compat/compat.h:787:0: warning: "call_rcu" redefined
 #define call_rcu call_rcu_bh

In file included from ./include/linux/srcu.h:33:0,
                 from ./include/linux/notifier.h:16,
                 from ./include/linux/memory_hotplug.h:7,
                 from ./include/linux/mmzone.h:782,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /home/pi/WireGuard/src/compat/compat.h:325,
                 from <command-line>:0:
./include/linux/rcupdate.h:55:0: note: this is the location of the previous definition
 #define call_rcu call_rcu_sched

  CC [M]  /home/pi/WireGuard/src/noise.o
In file included from <command-line>:0:0:
/home/pi/WireGuard/src/compat/compat.h:787:0: warning: "call_rcu" redefined
 #define call_rcu call_rcu_bh

In file included from ./include/linux/srcu.h:33:0,
                 from ./include/linux/notifier.h:16,
                 from ./include/linux/memory_hotplug.h:7,
                 from ./include/linux/mmzone.h:782,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /home/pi/WireGuard/src/compat/compat.h:325,
                 from <command-line>:0:
./include/linux/rcupdate.h:55:0: note: this is the location of the previous definition
 #define call_rcu call_rcu_sched

  CC [M]  /home/pi/WireGuard/src/device.o
In file included from <command-line>:0:0:
/home/pi/WireGuard/src/compat/compat.h:787:0: warning: "call_rcu" redefined
 #define call_rcu call_rcu_bh

In file included from ./include/linux/srcu.h:33:0,
                 from ./include/linux/notifier.h:16,
                 from ./include/linux/memory_hotplug.h:7,
                 from ./include/linux/mmzone.h:782,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /home/pi/WireGuard/src/compat/compat.h:325,
                 from <command-line>:0:
./include/linux/rcupdate.h:55:0: note: this is the location of the previous definition
 #define call_rcu call_rcu_sched

  CC [M]  /home/pi/WireGuard/src/peer.o
In file included from <command-line>:0:0:
/home/pi/WireGuard/src/compat/compat.h:787:0: warning: "call_rcu" redefined
 #define call_rcu call_rcu_bh

In file included from ./include/linux/srcu.h:33:0,
                 from ./include/linux/notifier.h:16,
                 from ./include/linux/memory_hotplug.h:7,
                 from ./include/linux/mmzone.h:782,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /home/pi/WireGuard/src/compat/compat.h:325,
                 from <command-line>:0:
./include/linux/rcupdate.h:55:0: note: this is the location of the previous definition
 #define call_rcu call_rcu_sched

[... 117 lines cut ...]
In file included from ./include/linux/srcu.h:33:0,
                 from ./include/linux/notifier.h:16,
                 from ./include/linux/memory_hotplug.h:7,
                 from ./include/linux/mmzone.h:782,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /home/pi/WireGuard/src/compat/compat.h:325,
                 from <command-line>:0:
./include/linux/rcupdate.h:55:0: note: this is the location of the previous definition
 #define call_rcu call_rcu_sched

/home/pi/WireGuard/src/ratelimiter.c: In function ‘entry_uninit’:
/home/pi/WireGuard/src/ratelimiter.c:60:2: error: implicit declaration of function ‘call_rcu’ [-Werror=implicit-function-declaration]
  call_rcu(&entry->rcu, entry_free);
  ^~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:326: recipe for target '/home/pi/WireGuard/src/ratelimiter.o' failed
make[2]: *** [/home/pi/WireGuard/src/ratelimiter.o] Error 1
Makefile:1535: recipe for target '_module_/home/pi/WireGuard/src' failed
make[1]: *** [_module_/home/pi/WireGuard/src] Error 2
Makefile:37: recipe for target 'module' failed
make: *** [module] Error 2
`

I tied after update && upgrade and before. Both gave the same result. Any help will be apperciated.

adrianmihalko, 2019-03-17:
Please connect to Wireguard community on #Wireguard (IRC, Freenode), here you can also find the developer of Wireguard. If you found the error, please post here.

modidep, 2019-03-17:
Will do Adrian. Thanks

lubbo, 2019-03-17:
Same compilation problem for me!

MichaIng, 2019-03-25:
Are you guys trying to compile it from source or via wireguard-dkms from Debian unstable repo?

AFAIK it is a known issue the Debian unstable packages work only on ARMv7+, thus not on RPi1 nor on RPi Zero.

You need to compile from source then, e.g.:
```
sudo apt install libmnl-dev build-essential git raspberrypi-kernel-headers qrencode
git clone https://git.zx2c4.com/WireGuard
cd WireGuard/src
make
sudo make install
sudo modinfo wireguard
sudo modprobe wireguard
```

lubbo, 2019-03-26:
From https://git.zx2c4.com/WireGuard/snapshot/WireGuard-0.0.20181218.tar.xz

lubbo, 2019-03-26:
> Are you guys trying to compile it from source or via wireguard-dkms from Debian unstable repo?
>
> AFAIK it is a known issue the Debian unstable packages work only on ARMv7+, thus not on RPi1 nor on [...]

$ make
  CC [M]  /home/pi/WireGuard/src/main.o
  CC [M]  /home/pi/WireGuard/src/noise.o
  CC [M]  /home/pi/WireGuard/src/device.o
/home/pi/WireGuard/src/device.c: In function 'wg_xmit':
/home/pi/WireGuard/src/device.c:165:3: error: implicit declaration of function 'skb_mark_not_on_list' [-Werror=implicit-function-declaration]
   skb_mark_not_on_list(skb);
   ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:326: recipe for target '/home/pi/WireGuard/src/device.o' failed
make[2]: *** [/home/pi/WireGuard/src/device.o] Error 1
Makefile:1535: recipe for target '_module_/home/pi/WireGuard/src' failed
make[1]: *** [_module_/home/pi/WireGuard/src] Error 2
Makefile:37: recipe for target 'module' failed
make: *** [module] Error 2

lubbo, 2019-03-26:
I was able to compile using snapshot https://git.zx2c4.com/WireGuard/snapshot/WireGuard-0.0.20181218.tar.xz

But the client cannot properly manage the handshake.
Already posted to the old-style mailing list but nobody help me ...

```
2019-03-20 09:49:03.715134: [APP] App version: 0.0.20190319 (5); Go backend version: 0.0.20181222
2019-03-20 14:53:14.699723: [APP] startActivation: Entering (tunnel: Home)
2019-03-20 14:53:14.728403: [APP] startActivation: Starting tunnel
2019-03-20 14:53:14.730296: [APP] startActivation: Success
2019-03-20 14:53:14.736005: [APP] Tunnel 'Home' connection status changed to 'connecting'
2019-03-20 14:53:14.939422: [NET] App version: 0.0.20190319 (5); Go backend version: 0.0.20181222
2019-03-20 14:53:14.939872: [NET] Starting tunnel from the app
2019-03-20 14:53:19.728990: [APP] Status update notification timeout for tunnel 'Home'. Tunnel status is now 'connecting'.
2019-03-20 14:53:19.971471: [NET] Tunnel interface is utun1
2019-03-20 14:53:19.972401: [NET] Attaching to interface
2019-03-20 14:53:19.973319: [NET] Routine: decryption worker - started
2019-03-20 14:53:19.973413: [NET] Routine: event worker - started
2019-03-20 14:53:19.973610: [NET] Routine: handshake worker - started
2019-03-20 14:53:19.973693: [NET] Routine: encryption worker - started
2019-03-20 14:53:19.973871: [NET] Routine: decryption worker - started
2019-03-20 14:53:19.973924: [NET] Routine: encryption worker - started
2019-03-20 14:53:19.974081: [NET] Routine: encryption worker - started
2019-03-20 14:53:19.974133: [NET] Routine: handshake worker - started
2019-03-20 14:53:19.974183: [NET] Routine: decryption worker - started
2019-03-20 14:53:19.974229: [NET] Routine: decryption worker - started
2019-03-20 14:53:19.974361: [NET] Routine: handshake worker - started
2019-03-20 14:53:19.974423: [NET] Routine: handshake worker - started
2019-03-20 14:53:19.974579: [NET] Routine: TUN reader - started
2019-03-20 14:53:19.974704: [NET] Routine: encryption worker - started
2019-03-20 14:53:19.974859: [NET] UAPI: Updating private key
2019-03-20 14:53:19.975040: [NET] UAPI: Removing all peers
2019-03-20 14:53:19.975092: [NET] UAPI: Transition to peer configuration
2019-03-20 14:53:19.975482: [NET] peer(Jn7u…NMyQ) - UAPI: Created
2019-03-20 14:53:19.975551: [NET] peer(Jn7u…NMyQ) - UAPI: Updating endpoint
2019-03-20 14:53:19.975906: [NET] peer(Jn7u…NMyQ) - UAPI: Updating persistent keepalive interval
2019-03-20 14:53:19.976108: [NET] peer(Jn7u…NMyQ) - UAPI: Removing all allowedips
2019-03-20 14:53:19.976157: [NET] peer(Jn7u…NMyQ) - UAPI: Adding allowedip
2019-03-20 14:53:19.976324: [NET] peer(Jn7u…NMyQ) - UAPI: Adding allowedip
2019-03-20 14:53:19.976811: [NET] Routine: receive incoming IPv4 - started
2019-03-20 14:53:19.977860: [NET] Routine: receive incoming IPv6 - started
2019-03-20 14:53:19.977988: [NET] UDP bind has been updated
2019-03-20 14:53:19.978138: [NET] peer(Jn7u…NMyQ) - Starting...
2019-03-20 14:53:19.978286: [NET] peer(Jn7u…NMyQ) - Routine: nonce worker - started
2019-03-20 14:53:19.978485: [NET] peer(Jn7u…NMyQ) - Routine: sequential receiver - started
2019-03-20 14:53:19.978537: [NET] peer(Jn7u…NMyQ) - Routine: sequential sender - started
2019-03-20 14:53:19.978637: [NET] peer(Jn7u…NMyQ) - Sending keepalive packet
2019-03-20 14:53:19.978707: [NET] Device started
2019-03-20 14:53:19.978719: [NET] peer(Jn7u…NMyQ) - Sending handshake initiation
2019-03-20 14:53:19.979785: [NET] peer(Jn7u…NMyQ) - Awaiting keypair
2019-03-20 14:53:19.980234: [APP] Tunnel 'Home' connection status changed to 'connected'
2019-03-20 14:53:20.242577: [NET] peer(Jn7u…NMyQ) - Received handshake response
2019-03-20 14:53:20.242843: [NET] peer(Jn7u…NMyQ) - Obtained awaited keypair
2019-03-20 14:53:23.025628: [NET] UAPI: Processing get operation
2019-03-20 14:53:24.013681: [NET] UAPI: Processing get operation
2019-03-20 14:53:24.729239: [APP] Status update notification timeout for tunnel 'Home'. Tunnel status is now 'connected'.
2019-03-20 14:53:24.729967: [NET] UAPI: Processing get operation
2019-03-20 14:53:25.731556: [NET] UAPI: Processing get operation
2019-03-20 14:53:26.730482: [NET] UAPI: Processing get operation
2019-03-20 14:53:27.730208: [NET] UAPI: Processing get operation
2019-03-20 14:53:28.730320: [NET] UAPI: Processing get operation
2019-03-20 14:53:29.731715: [NET] UAPI: Processing get operation
2019-03-20 14:53:30.730575: [NET] UAPI: Processing get operation
2019-03-20 14:53:31.731652: [NET] UAPI: Processing get operation
2019-03-20 14:53:32.731655: [NET] UAPI: Processing get operation
2019-03-20 14:53:33.731668: [NET] UAPI: Processing get operation
2019-03-20 14:53:34.731727: [NET] UAPI: Processing get operation
2019-03-20 14:53:35.246810: [NET] peer(Jn7u…NMyQ) - Retrying handshake because we stopped hearing back after 15 seconds
2019-03-20 14:53:35.247047: [NET] peer(Jn7u…NMyQ) - Sending handshake initiation
2019-03-20 14:53:35.490732: [NET] peer(Jn7u…NMyQ) - Received handshake response
[... 800 lines cut ...]
2019-03-26 09:37:57.906333: [NET] UAPI: Processing get operation
2019-03-26 09:37:58.906396: [NET] UAPI: Processing get operation
2019-03-26 09:37:59.906738: [NET] UAPI: Processing get operation
2019-03-26 09:38:00.907656: [NET] UAPI: Processing get operation
2019-03-26 09:38:01.907437: [NET] UAPI: Processing get operation
2019-03-26 09:38:02.907579: [NET] UAPI: Processing get operation
2019-03-26 09:38:03.907591: [NET] UAPI: Processing get operation
[... cut ...]
2019-03-26 09:38:12.295233: [NET] peer(Jn7u…NMyQ) - Sending handshake initiation
2019-03-26 09:38:12.369944: [NET] peer(Jn7u…NMyQ) - Received handshake response
2019-03-26 09:38:12.370188: [NET] peer(Jn7u…NMyQ) - Sending keepalive packet
2019-03-26 09:38:12.907647: [NET] UAPI: Processing get operation
2019-03-26 09:38:13.906729: [NET] UAPI: Processing get operation
2019-03-26 09:38:14.907178: [NET] UAPI: Processing get operation
2019-03-26 09:38:16.090998: [NET] UAPI: Processing get operation
2019-03-26 09:38:16.907014: [NET] UAPI: Processing get operation

adrianmihalko, 2019-03-26:
Please join to Wireguard channel on IRC, there you can get help easier.

modidep, 2019-03-26:
I tried from source. Nobody replied on IRC.

lubbo, 2019-03-26:
OMG ... IRC and mailing list ... isn’t it a bit strange the usage of those vintage support channels for a product that aims to innovate the security?

adrianmihalko, 2019-03-26:
Emma, no?

KardinalReusen, 2020-07-30:
I have the same problem with my raspberry pi 1 model B. As modidep I asked for help on the #wireguard IRC channel without getting any reply. Is there any solution for this compile error problem?