adrianmihalko/raspberrypiwireguard #20: gpg signature can't be verified.

plainas, 2019-04-09:
Hey. I did get wireguard installed on a raspberry pi 3 using these instructions. However, the packages are not verified. It appears that the instructions to import the key need to be fixed/updated.

This is the error I got.

```Reading package lists... Done
W: GPG error: http://deb.debian.org/debian unstable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
W: The repository 'http://deb.debian.org/debian unstable InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

jm33-m0, 2019-04-13:
as the error message indicates, you should import the two public keys as well:
```
apt-key adv --keyserver   keyserver.ubuntu.com --recv-keys 7638D0442B90D010
apt-key adv --keyserver   keyserver.ubuntu.com --recv-keys 04EE7237B7D453EC
```

jm33-m0, 2019-04-13:
created a pr to fix #20:
https://github.com/adrianmihalko/raspberrypiwireguard/pull/21

plainas, 2019-04-23:
Closing in favor of #21

closed 2019-04-23 by plainas

CanntAim, 2019-08-06:
These are keys from debian archive key ring? It seems like the better idea is to just run:

```
sudo apt-get install debian-archive-keyring
sudo apt-get install debian-keyring
```

farrukh43, 2020-07-23:
apt-key adv --keyserver hkp://keys.gnupg.net --recv-keys 7638D0442B90D010
apt-key adv --keyserver hkp://keys.gnupg.net --recv-keys  04EE7237B7D453EC

wahabshah, 2021-04-22:
I had an old container which had debain stretch with `debian-archive-keyring` at version `2017.5` and suddenly it stopped working as with the above errors caused because of not having `buster`  keys although i had container with stretch.
But then did manually update of `debian-archive-keyring` to version `2017.5+deb9u1` after logging into container and it worked:-

```sh
sudo apt update && sudo apt-get install --only-upgrade debian-archive-keyring
```

and then checking it in apt-key :-
```sh
sudo apt-key list
```
It showed then `buster` keys in the list and could update