adrianmihalko/raspberrypiwireguard #26: connected but no internet access

firepmi, 2019-08-22:
I installed wireguard on digitalocean ubuntu according to the guideline. 
the client is connected, but no internet access.
what did I do wrong?

ccebrecos, 2019-09-14:
Same here. Installed server on ArchLinux; I can access to the server but no to internet, seems it's not forwarding the packets...

RubenNL, 2019-09-14:
I just solved my problem(other way of installing, but may solve it for you) by enabling ipv4 forwarding on the server.

Currently I am unable to search it myself, but Google "enable ipv4 forwarding Linux". I think that may solve your issue

ThePigPeppa, 2019-09-20:
**Automated Script:**

```
https://github.com/complexorganizations/wireguard-install
```

ccebrecos, 2019-09-30:
Hi !
ipv4 forwarding was already enabled on the server and iptables properly configured.
Don't think the problem was there..

IssueFindings, 2019-12-03:
Hello, I have the same issue. I already check the ipv4 forwarding on my server (net.ipv4.ip_forward = 1). Neevertheless my server can't access to Internet. I think is linked with Wireguard configuration but I don't see what is wrong. Could you help me ? Have a nice day.

My file configuration (/etc/wireguard/wg0.conf) is :
`[Interface]
Address = 10.9.0.1/24 #The local ip given by Internet box is 192.168.X.X
ListenPort = 51820
DNS = local IP of my server (192.168.Y.Y)
PrivateKey = server_privatekey
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -A FORWARD -o %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth1 -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -D FORWARD -o %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth1 -j MASQUERADE
[Peer]
#Peer-1
PublicKey = peer1_publickey
AllowedIPs = 10.9.0.2/32
#PersistentkeepAlive = 60`

IssueFindings, 2019-12-03:
Issue solved, I put 1.1.1.1 as IP for DNS option in the file below. How I can check that everything go through WireGuard ?
Have a nice day.

ccebrecos, 2019-12-04:
Hi there!
In my case the issue was a rule on ip rule; fixed just removing that rule:
`PostUp = ... ; ip rule delete table <ListenPort>`

AshleyDumaine, 2019-12-05:
I'm also running into this issue. Wireguard was working for me earlier this year and nothing changed with the configuration for the server or client, but it now I have no internet access even though it's showing the client is connecting to the server. I did notice some errors are showing up on the wg interface:
```
$ ifconfig wg0
wg0: flags=209<UP,POINTOPOINT,RUNNING,NOARP>  mtu 1420
        inet 10.0.0.1  netmask 255.255.255.0  destination 10.0.0.1
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 219  bytes 34236 (33.4 KiB)
        RX errors 21  dropped 0  overruns 0  frame 21
        TX packets 328  bytes 104396 (101.9 KiB)
        TX errors 0  dropped 1 overruns 0  carrier 0  collisions 0
```

```
wg show
interface: wg0
<snip>
  latest handshake: 41 seconds ago
  transfer: 33.43 KiB received, 101.95 KiB sent
```
Haven't figured out a fix yet unfortunately.