adrianmihalko/raspberrypiwireguard #36: activate Fail2Ban for WireGuard

IssueFindings, 2019-12-25:
Hello,
How I must do to activate Fail2Ban to check WireGuard activities ? Normally, I use the log file but I don't know where is the log file for logins failed. Someone can help me ?
Have a nice day.

Prajwal-Koirala, 2020-03-02:
fail2ban is not required for wireguard since if they don't authenticate with the right key the server never responds

IssueFindings, 2020-06-24:
Hello,
Many thanks for your answer. Have a nice day.

closed 2020-06-24 by IssueFindings

c33s, 2020-12-16:
shouldn't the server still block the ip via firewall? also fail2ban can alert if some ip tries to connect to the server to often or trigger other countermeasures or block the ip on other hosts.

raytracy, 2021-07-05:
> shouldn't the server still block the ip via firewall? also fail2ban can alert if some ip tries to connect to the server to often or trigger other countermeasures or block the ip on other hosts.

I also vote for the wireguard should log connection failure event for security audit and CSIRT response.

effeffe, 2023-01-13:
Sorry for necrobumping, 
you can enable logging on journalctl via `echo module wireguard +p > /sys/kernel/debug/dynamic_debug/control`, or at boot via `wireguard.dyndbg=+p`. Then write an appropriate fail2ban rule, which could for instance filter the following request:
`wireguard: serverino: No peer has allowed IPs matching 239.255.255.250` or any other similar request.