adrianmihalko/raspberrypiwireguard #47: Raspberry Pi 1, Zero and ZeroW wiki page needs to be updated

manualmanul, 2020-09-03:
Hi,

Please update the [wiki/Install-WireGuard-on-Raspberry-Pi-1,-2-(not-v1.2),-Zero,-Zero-W](https://github.com/adrianmihalko/raspberrypiwireguard/wiki/Install-WireGuard-on-Raspberry-Pi-1,-2-(not-v1.2),-Zero,-Zero-W) wiki page to point to the correct WireGuard git repository: <https://git.zx2c4.com/wireguard-linux-compat/>. Currently mentioned git repository is outdated and causes a compilation error.

I've found this information in the #46 issue, which was unfortunately closed by the author.

manualmanul, 2020-09-03:
Also, a section for compiling wireguard-tools should be added, the repository for which is <https://git.zx2c4.com/wireguard-tools>

adrianmihalko, 2020-09-03:
@manualmanul any pull requests for changes are welcomed.

manualmanul, 2020-09-04:
@adrianmihalko GitHub wiki pages don't support pull requests or editing by non-contributors. Unless you make me a contributor, I can't do anything other than create an issue.

saudiqbal, 2021-04-22:
This is what worked for my PI 1

```
pi@raspberrypi:~ $ git clone https://git.zx2c4.com/wireguard-tools
pi@raspberrypi:~ $ cd wireguard-tools/src
pi@raspberrypi:~/WireGuard/src $ make
pi@raspberrypi:~/WireGuard/src $ sudo make install
```